smcneece/ha-esunpower #4: Possible bug with the newest update that enabled nightime polling:

dovy6, 2025-09-17:
See attached image:

[image]

It's saying it's using cached data. I enabled debugging messages temporarily to try to figure this out. I'm not sure, but I think it's actually polling every 600 seconds as I have it set to for days, and not every 3600 as I have it for nights. Do you know why it says it's using cached data?

Incidentally, aren't new notifications supposed to wipe away the old? Or is that only for mobile_app notifications?

Thanks again
Dovy

smcneece, 2025-09-17:
Ok I think it's fixed in my test version. Will push it up shortly after a bit more testing. Also enhanced one of the debug messages, which is how I tracked the issue down.  I have had some stability issues with my PVS a few months ago, so I built an extensive notification system to save me time (I'm lazy) from digging through logs, it just pops up what is going on.  My test VM constantly has debug mode, and my production Home Assistant just has replace status" as I like to know it's working. 

[image]

smcneece, 2025-09-19:
Ok just pushed 2025.9.3, this should hopefully fix the issues, has been running smooth for me for 30+ hours, and several day/night transitions.